skelgaard/homeassistant-apsystems #3: Since today no metrics

synack666, 2022-02-09:
Since today there a no metrics in HA, i noticed that there is a captcha on apsystemsema server in browser, this is a change from yesterday.

Log HA:

Logger: homeassistant.helpers.entity
Source: custom_components/apsystems/sensor.py:241
Integration: apsystems (documentation)
First occurred: 07:32:28 (100 occurrences)
Last logged: 09:07:34

    Update for sensor.apsystems_energy_day fails
    Update for sensor.apsystems_energy_latest fails
    Update for sensor.apsystems_power_max_day fails
    Update for sensor.apsystems_power_latest fails
    Update for sensor.apsystems_date fails

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 487, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 691, in async_device_update
    raise exc
  File "/config/custom_components/apsystems/sensor.py", line 146, in async_update
    ap_data = await self._fetcher.data()
  File "/config/custom_components/apsystems/sensor.py", line 252, in data
    await self.run()
  File "/config/custom_components/apsystems/sensor.py", line 241, in run
    self.cache = result_data.json()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 910, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/site-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.9/site-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 5 column 1 (char 4)

bulldog39, 2022-02-09:
Same here I can not see any sensor

skelgaard, 2022-02-09:
Looking into it and seems to be a fault at their end... but looking on into it....
We are so sorry, EMA has encountered an error.<br/>click [OK] to email us and we will deal with it within 48 hours. Please re-login 
your system.

bulldog39, 2022-02-09:
Where to press OK? yesterday it seems to work
apsystems: Error on device update!

skelgaard, 2022-02-09:
well the short answer is, that this is now dead... i can't login without a code and i can't get the code from the image in any easy way.... i will mail them and ask if there is any other way to login and get that info from their site, so that home assistant still can get the data... but else i can't see another way at current....

Shadyville08, 2022-02-09:
> well the short answer is, that this is now dead... i can't login without a code and i can't get the code from the image in any easy way.... i will mail them and ask if there is any other way to [...]

I’m sorry to hear that. This was such a great component. Let’s hope apsystems shows some interest and tries to help us.

skelgaard, 2022-02-09:
well i have found a new way to login, now i just need to get it to work... i'm not a python coder, so i have to learn the python ways todo it... but working on it...

synack666, 2022-02-10:
thanks skelgaard for invest, i am familar with shell script but not python and try to follow your work an try to understand

skelgaard, 2022-02-19:
i have gotten it work now, just need homeassistant to play along :) still working on it

skelgaard, 2022-02-19:
i have data on my graph again... i will make the changes to the config and make sure all that working and expect a new release later today that works again

skelgaard, 2022-02-19:
version 1.0.5 has ben pushed with a fix for this.. please say if it works for you too...

synack666, 2022-02-19:
Hi, 1.0.5 failed:
 Logger: homeassistant.config
Source: config.py:454
First occurred: 14:09:17 (1 occurrences)
Last logged: 14:09:17
Invalid config for [sensor.apsystems]: [username] is an invalid option for [sensor.apsystems]. Check: sensor.apsystems->username. (See /config/configuration.yaml, line 67). Please check the docs at https://github.com/skelgaard/homeassistant-apsystems/blob/master/README.md

skelgaard, 2022-02-19:
disable your config in your yaml file, restart, enable config in yaml with authid ,restart working

skelgaard, 2022-02-19:
if still have problems, i'm on the home assistant discord for help :)

synack666, 2022-02-19:
Yeah!
Have not noticed to change the config params, now it works!
Thanks a lot, where ist the "spent a coffee" button?

closed 2022-02-19 by synack666

skelgaard, 2022-02-19:
its comming :)

synack666, 2022-02-20:
"Not working at current" from the main page is outdated.

synack666, 2022-02-20:
and, today, the captcha on apsystemsema.com is gone

skelgaard, 2022-02-20:
removed :)

skelgaard, 2022-02-20:
i have seen its gone, but i'm guessing that it will return at some point... so i'm not going to change it back to use username/password... but i may make it an option that can be selected to use that or the authid at some point.

synack666, 2022-02-20:
yes, i guess they test and check the acceptance, authid would the save way if they dont change that