skelgaard/homeassistant-apsystems #4: Error on device update

bulldog39, 2022-02-20:
Hello,

i have put in the new details in the config but I get an error. the old integration before the captcha works now again

synack666, 2022-02-20:
New version works here as expected. Config as discribed on main page.
Did you marked "allow visitor" on apsystemsema.com?
sensor:
  - platform: apsystems
    authId: apsystemsema_user
    systemId: apsystemsema_system_id
    ecuId: apsystemsema_ecu_id
    sunset: off

andrebbruno, 2022-02-20:
> New version works here as expected. Config as discribed on main page. Did you marked "allow visitor" on apsystemsema.com? sensor:
> 
> * platform: apsystems
>   authId: apsystemsema_user
>   systemId: apsystemsema_system_id
>   ecuId: apsystemsema_ecu_id
>   sunset: off

Same here!

bulldog39, 2022-02-20:
yes I have allowed visitors. 

Logger: homeassistant.components.sensor
Source: custom_components/apsystems/sensor.py:227 
Integration: Sensor (documentation, issues) 
First occurred: 18:20:52 (5 occurrences) 
Last logged: 18:21:04

apsystems: Error on device update!
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 910, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/site-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.9/site-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 5 column 1 (char 8)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 431, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 718, in async_device_update
    await task
  File "/config/custom_components/apsystems/sensor.py", line 142, in async_update
    ap_data = await self._fetcher.data()
  File "/config/custom_components/apsystems/sensor.py", line 238, in data
    await self.run()
  File "/config/custom_components/apsystems/sensor.py", line 227, in run
    self.cache = result_data.json()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 917, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: [Errno Expecting value] 

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Frameset//EN">
<html>
<head>
<title>apsystemsema</title>
<link rel="shortcut icon" href="https://apsystemsema.com:443/ema/imagesDy-1.0/login/logo.png" />
<link rel="stylesheet" type="text/css" href="https://apsystemsema.com:443/ema/cssDy-1.0/opt/fonts/fontController.css">
<link rel="stylesheet" type="text/css" href="https://apsystemsema.com:443/ema/cssDy-1.0/opt/opt.style.css">
<link rel="stylesheet" type="text/css" href="https://apsystemsema.com:443/ema/cssDy-1.0/opt/opt.ui.css">
<style>
body {
	margin: 0;
	padding: 0;
	font-size: 12px;
    min-height: 250px;
}

.flex {
	display: flex;
	justify-content: center;
	align-items: center;
	flex-wrap: wrap;
	flex-direction: column;
}

.title-text {
	font-size: 40px;
	font-weight: bold;
	text-align: center;
	width: 60%;
	margin-bottom: 3%;
}

.info-text {
	font-size: 20px;
	font-weight: bold;
	text-align: center;
	width: 60%;
	margin-bottom: 4%;
	line-height: 32px;
}

.back-text {
	font-size: 15px;
	font-weight: bold;
	text-align: center;
	width: 60%;
	margin-bottom: 2%;
}

.btn {
	opacity: 0.8;
	width: 80px;
}
</style>

</head>

<body>
	<div class="flex" style="height: 100%;">
		<div class="title-text">Note</div>
		<div class="info-text">We are so sorry, EMA has encountered an error.<br/>click [OK] to email us and we will deal with it within 48 hours. Please re-login your system.</div>
		
		<div>
			<input type="button" class="submit-small btn" value="OK" onclick="btnClick()" />
		</div>
		<div style="display: none">
		  	
		</div>
	</div>
	<script type="text/javascript">
//		var btn = document.getElementById("second");
//		var i = 30;
//		var timer = setInterval(function(){
//			i--;
//			btn.innerHTML = i;
//			if(i==0){
//				btnClick();
//			}
//		},1000);
		function getRootPath(id){
	       var path=window.document.location.href;
	       var pathName=window.document.location.pathname;
	       var pos=path.indexOf(pathName);
	       var localhostPath=path.substring(0,pos);
	       var projectName=pathName.substring(0,pathName.substr(1).indexOf('/')+1);
	       return(localhostPath+projectName+'/');
	     }
		function btnClick(){
			window.top.location.href= getRootPath()+"index.action";
		}

//		function copyClick(id) {
//			var Url2 = document.getElementById(id);
//            Url2.select(); // 选择对象 
//            document.execCommand("Copy"); // 执行浏览器bai复制命du令 
//		}
	</script>
</body>
</html>
: 8

bulldog39, 2022-02-20:
ok sorry, I have changed to sunset on and now I see the sensors.

synack666, 2022-02-21:
:-) ok, so you could close the issue, right?

bulldog39, 2022-02-21:
so after some restarts the sensors are gone away, I still have no sensors. I have reinstall and remove it but still no luck

synack666, 2022-02-22:
After restart in "dark time" i have no sensor too, have to restart when solar give energie, "sun time", everything is ok. On apsystemsema you could see in dark time, "no data".

skelgaard, 2022-02-25:
i have seen that problem my selves and its on the list of things i need to look into and fix at some point, but i just do a restart when sun is up and all is good :)

closed 2022-02-25 by skelgaard

Machiel92, 2022-03-08:
Hmm.. I still have the same error as mentioned first:
```
Logger: homeassistant.components.sensor
Source: custom_components/apsystems/sensor.py:227
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 17:05:28 (5 occurrences)
Last logged: 17:05:33

apsystems: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 431, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 633, in async_device_update
    await task
  File "/config/custom_components/apsystems/sensor.py", line 142, in async_update
    ap_data = await self._fetcher.data()
  File "/config/custom_components/apsystems/sensor.py", line 238, in data
    await self.run()
  File "/config/custom_components/apsystems/sensor.py", line 227, in run
    self.cache = result_data.json()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 900, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/site-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.9/site-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 5 column 1 (char 8)
```

Somehow I just can't get it to working.
Do you guys got it working at the moment?

synack666, 2022-03-08:
Yes, runs fine without any problem.

Machiel92, 2022-03-08:
Nevermind, found the issue made a booboo on the settings!

Works perfect!

Machiel92, 2022-03-08:
and did you got them included on the energy page :)?

synack666, 2022-03-08:
fore sure, see the open issue, there is the solution