skelgaard/homeassistant-apsystems #12: Sensors updates only once an hour.

zylxpl, 2022-04-12:
Hi everyone,
 Sensor updates only once per hour. Because of that it is hard to do anything useful with the data. Can we somehow increase pooling rate?

synack666, 2022-04-12:
right, this took me to look around, found this and feel happy:

https://github.com/ksheumaker/homeassistant-apsystems_ecur

zylxpl, 2022-04-13:
@synack666 sure, but this is much harder to implement for me, as the HA server is not connected via LAN directly to ECU.

synack666, 2022-04-13:
No problem, same here, my HA server is connected via Lan to my router, ECU ist connected via WLAN to my router, works fine.

synack666, 2022-04-13:
they have to be in the same network, thats all

zylxpl, 2022-04-13:
@synack666 connection method is irrelevant, i meant that my HA server and ecu are not on the same local network and are not evn in the same location.  To get this work i would need to create some vpn between two locations.

synack666, 2022-04-13:
right, now i understand :-)

skelgaard, 2022-04-14:
it does pull more than once per hour, but the energy dashboard doesn't update more than once per hour... so it matters where you want it to show ?
If you need live data, you can't do it with this, as the apsystem doesn't give data that way. its set to request every 20 sec, so it can be lowered, but shouldn't be much better..
if you want live data then as @synack666 says that addon is better (though i can't get my ECUR to give me that interface) or get a shelly 3em and add that to the outputs from the solar and messure with that live what it produces

closed 2022-04-14 by skelgaard

zylxpl, 2022-04-16:
@skelgaard I'm not referring to Energy dashboard, don't using it anyway. What is a problem for me is that apsystems sensors update once hour on even less. Here is a sample readout of apsystems_energy_day:

[image]

As you can see it only updates once a hour. Around 13.30 i restarted HA and  got the current value for that time. So the data is accessible.

didigrand, 2022-10-20:
@skelgaard @ylxpl I have the exact same issue...
The apsystems_energy_day updates only once an hour.
But when I look at the log, the automation run every 5 minutes.