skelgaard/homeassistant-apsystems #13: Error on device update

freaksdotcom, 2022-08-05:
The integration is failing with a TypeError:

```
2022-08-05 02:31:04.424 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform apsystems is taking over 10 seconds.
2022-08-05 02:31:07.258 ERROR (MainThread) [homeassistant.components.sensor] apsystems: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 477, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 702, in async_device_update
await task
File "/config/custom_components/apsystems/sensor.py", line 142, in async_update
ap_data = await self._fetcher.data()
File "/config/custom_components/apsystems/sensor.py", line 246, in data
timestamp_event = int(self.cache['time'][-1]) + offset_hours # apsystems have 8h delayed in timestamp from UTC
TypeError: 'NoneType' object is not subscriptable
```

skelgaard, 2022-08-05:
do you still have this error ? i selves seen some wierd errors in the nite time, but works fine in the day again

freaksdotcom, 2022-08-05:
It works when the panels are active, but I also suspect that the problem will come up again if HA restarts after sunset.

The bug was introduced in https://github.com/skelgaard/homeassistant-apsystems/commit/e7cfdba2bb053ddb4e002b50a0ba050743fb1ceb on line 255-256. If the panels are not producing power,  `self.cache` will never be populated. Line 260 tries to subscript it, causing the `TypeError`

skelgaard, 2022-08-05:
well thats around the same i came to when trying to find the bug... it also has something todo with that offset in there, just haven't had the time to figure out how to fix it and as it works when there is solar produced, i have not looked much more into it

skelgaard, 2022-08-17:
that works for me :) so do tell if it doesn't for you

closed 2022-10-09 by skelgaard