skelgaard/homeassistant-apsystems #14: Can add entity apsystems_enrgy|_day to dasboard, but not in the energy tab

odiejakkes, 2022-08-18:
thanks a lot for this fork, it works indeed,and although thank you for your message, i found your fork
myselves after some digging 👍 
the entitys show up when making a card, have them in my dashboard besides the ECU-3 energy today.
because i own a ECU-R and this damned ECU-3
strangely  directly after installing your integration, it showed up as second available energy source in the energy tab
but after a couple of minutes the apsystem_energy_day was replaced by the ECU-TODAY_ENERGY entity so it was double 
after deleting this double, the entities apsystem do not show in the energy tab anymore

any clue why this happened ?

it would all be solved when support apsytems anweres my question whether a ECU-3 ID can be added to a ECU-R account, 
so both solar installations show up in 1 emea account, now they are in seperate accounts....
but the support of apsystems is very minimal, they don't read questions and answer slowly

i'll inspect both sensor.py if there are conflicting identical defenitions of a string
maybe that's why the ECU-R entitys took over your apsystems entity

odiejakkes, 2022-08-18:
ah. maybe the sensor.py misses the device class to be able to show up in the energy tab ?
this is from the sensor.py from the ECU-R integration 👍 

from homeassistant.const import (
    DEVICE_CLASS_TEMPERATURE,
    DEVICE_CLASS_POWER,
    DEVICE_CLASS_ENERGY,
    DEVICE_CLASS_VOLTAGE,
    DEVICE_CLASS_ENERGY,
    ENERGY_KILO_WATT_HOUR,
    POWER_WATT,
    ELECTRIC_POTENTIAL_VOLT,
    TEMP_CELSIUS,
    PERCENTAGE,
    FREQUENCY_HERTZ
)

skelgaard, 2022-08-18:
i don't know the ecu-3 and not much into python... i just forked the other one and fixed the few errors i have encountered to be able to use it my selves...
if you have any idea whats needs adding to get it to work, i will gladly add it to the source :)

skelgaard, 2022-11-21:
new release is out for it, you should be able to get data again

closed 2022-11-21 by skelgaard

odiejakkes, 2022-11-23:
Thanks, i'll give it a try….

odiejakkes, 2023-02-27:
Finaly got the time, and yes it's working now as a sensor, thanks a lot for
your
now i need to find how i can make an integration of it, so i can add the
KWH_day to my
energy dashboard, it now lacks a unique ID
seen others with the same issue, so hopefully i can find the info on their
replies

skelgaard, 2023-02-27:
Nice 👍 i look forward for a PR from you with your changes so that they can be reviewed and merged in :)