skelgaard/homeassistant-apsystems #16: No Unique ID

perco15, 2022-11-06:
Hello :) 

I'm just trying to install your component in my HASS system, but i have a problem, 

I find all information needed and put it in configuration.yaml, but when i go in integration i have a message : 

[image]

Can you help me to fix it ?

Thanks a lot :)

skelgaard, 2022-11-06:
right now its not working at all, cause something has changed in the request... i have gotten connection again, though home assistant doesn't agree with me on how to do it, i will get it working soon again

andrebbruno, 2022-11-07:
Hey guys - I´m using this other method for gathering ECU data and it´s working quite well. 

https://github.com/ksheumaker/homeassistant-apsystems_ecur

hope that helps while this one is getting fixed.

skelgaard, 2022-11-07:
that requires the local webinterface of the ECU to be active... mine isn't so i can't even build anything like that or use it... thanx though.

skelgaard, 2022-11-21:
latest version gives data again. try update and post if problems... i will look into adding unique id to them, so you can change stuff on them

closed 2022-11-21 by skelgaard