skelgaard/homeassistant-apsystems #20: Energy Dashboard statistics?

RacerBas, 2022-12-05:
I'm trying to add the solar panels to the energy dasboard but it doesn't seem to be supported?

https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics

Is this possible perhaps?

skelgaard, 2022-12-14:
mine is in my dashboard... what are you missing ?

RacerBas, 2022-12-15:
on the built-in energy dashboard you cannot add it into the solar production category. It's just not listed
[image]

skelgaard, 2022-12-16:
ah i can se i have this in my configuration... i will look into if i can get that set by default in the sensors
homeassistant:
  customize_glob:
    sensor.apsystems_energy_day:
      device_class: energy
      state_class: total_increasing

Bittabola, 2023-05-26:
Hi, has this been resolved? I don’t see the APsystems sensors in the dropdown when I click Add Solar Production. Thanks!

icecreamfarmer, 2023-05-26:
> Hi, has this been resolved? I don’t see the APsystems sensors in the dropdown when I click Add Solar Production. Thanks!

For me the same. Have been trying different yaml configs but I'll keep getting errors that the device class is not valid.

icecreamfarmer, 2023-05-29:
> Hi, has this been resolved? I don’t see the APsystems sensors in the dropdown when I click Add Solar Production. Thanks!

I Think you need to add that line or device class somewhere in here:
/config/custom_components/apsystems/sensor.py

Bu i can't get it to work.

skelgaard, 2023-05-29:
have you tried the code in configuration.yaml ?
i have not found out how to add this into the code, so that code in configuration is not needed.

icecreamfarmer, 2023-05-29:
> have you tried the code in configuration.yaml ? i have not found out how to add this into the code, so that code in configuration is not needed.

Yes and keep getting errors like:
Invalid domain sensor.apsystems_energy_day

icecreamfarmer, 2023-05-30:
> have you tried the code in configuration.yaml ? i have not found out how to add this into the code, so that code in configuration is not needed.

Let's put it this way. What exactly did you put into the config.yaml and where?

Bittabola, 2023-05-30:
I added the following code to configutation.yaml in the config folder:

`sensor:
  - platform: apsystems
    authId: 2c9f934a884b9324018858f15XXXXXX
    systemId: 2c9f934a884b93240188587XXXXXX
    ecuId: 2162XXXXXXX
    sunset: off`

skelgaard, 2023-06-17:
Seems i have fixed this in the next version that i will proberly release tomorrow... just doing final testings of it

closed 2023-06-17 by skelgaard