skelgaard/homeassistant-apsystems #22: Little issues

Ganzhans, 2022-12-27:
Good news , the basics are working but a few problems / issues:

1)  i can't see the senor  "Energy total"   the other sensors are ok , how can i add this last one ?
2) i have only  real time  sennors,  hoe can i add a dasboard with history over the day / month / year as in the  "pull resquest"  tab
3) i have setup a full HA system for testing, the prodcution is running on docker, can i add it later en put the files in the  \config\custom_components\apsystem  folder an reboot HA ?

Tnx for reading

skelgaard, 2023-06-17:
Seems i have fixed this in the next version that i will proberly release tomorrow... just doing final testings of it

closed 2023-06-17 by skelgaard

Ganzhans, 2023-06-24:
it works good for now !!  I have one little issue but is not a big problem. later I come back an will explain the "problem"