skelgaard/homeassistant-apsystems #23: apsystems: Error on device update!

joshmtucker, 2023-01-24:
I'm experiencing an "error on device update" issue, and don't seem to be able to see any sensors added. I reviewed closed issues and other threads in hopes I could figure out the solution on my own, but unfortunately I wasn't able to. I confirmed all the variables in my configuration.yaml are correct, installed the latest version from GitHub (1.08), and confirmed I have data available for today in apsystemsema.com.

Below is the output from Logs. Thanks!

```
Logger: homeassistant.components.sensor
Source: custom_components/apsystems/sensor.py:228
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 1:04:25 PM (5 occurrences)
Last logged: 1:04:38 PM

apsystems: Error on device update!
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.10/site-packages/simplejson/decoder.py", line 372, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.10/site-packages/simplejson/decoder.py", line 402, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 5 column 1 (char 8)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 503, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 729, in async_device_update
    await task
  File "/config/custom_components/apsystems/sensor.py", line 142, in async_update
    ap_data = await self._fetcher.data()
  File "/config/custom_components/apsystems/sensor.py", line 239, in data
    await self.run()
  File "/config/custom_components/apsystems/sensor.py", line 228, in run
    self.cache = result_data.json()
  File "/usr/local/lib/python3.10/site-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 5 column 1 (char 8)
```

skelgaard, 2023-06-17:
I have fixed this in the next version that i will proberly release tomorrow... just doing final testings of it

closed 2023-06-17 by skelgaard

joshmtucker, 2023-06-25:
I updated to 1.09 and I still appear to be experiencing the same issue. Here's the log output:

```
apsystems: Error on device update!
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/simplejson/__init__.py", line 525, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.10/site-packages/simplejson/decoder.py", line 372, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.10/site-packages/simplejson/decoder.py", line 402, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
  File "/usr/local/lib/python3.10/site-packages/simplejson/scanner.py", line 79, in scan_once
    return _scan_once(string, idx)
  File "/usr/local/lib/python3.10/site-packages/simplejson/scanner.py", line 70, in _scan_once
    raise JSONDecodeError(errmsg, string, idx)
simplejson.errors.JSONDecodeError: Expecting value: line 5 column 1 (char 8)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 739, in async_device_update
    await task
  File "/config/custom_components/apsystems/sensor.py", line 209, in async_update
    ap_data = await self._fetcher.data()
  File "/config/custom_components/apsystems/sensor.py", line 313, in data
    await self.run()
  File "/config/custom_components/apsystems/sensor.py", line 301, in run
    self.cache = result_data.json()
  File "/usr/local/lib/python3.10/site-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 5 column 1 (char 8)
```

skelgaard, 2023-06-26:
please turn on debug and post a debug log and do this in a new issue please... if you don't wanna post the debug, then send it in a private to me or poke me on discord... i'm in the home assistant once