skelgaard/homeassistant-apsystems #25: Data updated at odd times

rohrsh, 2023-02-11:
e.g. here is a chart of my last week: 
[image]

Data appears to be updated ~2 times per day, and is unavailable for a window. (I have sunset: false)

I wonder if if has anything to do with timezones. I am in Australia, a positive UTC offset.

skelgaard, 2023-02-11:
now that looks wierd....
Homeassistant should run UTC time then offset with your timezone to show you the correct time.
If your home assistant doesn't have a timezone set, well then it is the reason.

You can also try and change this 
in custom_components\apsystems\sensor.py
there is two places where offset_hours is set, this is set as AP is in +7 hours, so if you are +11, try change those to 4 hours and restart homeassistant... if that works, its cause of timezone problem and as far as i know then, its cause your timezone is set wrongly... but it can also be wrong fetched data.... i will check up on that.

skelgaard, 2023-02-11:
do your data look fine in the apsystem website ?
https://apsystemsema.com/ema/index.action

rohrsh, 2023-02-11:
Home Assistant time

[image]

When I click on that index. action link above it always takes me to APS login screen, even after I am logged in. 

Here's my main page. Can see data and browse history. You can see it's some Aussie brand (ZEN) wrapped around the APS Api. Perhaps that's the problem. 

[image]

skelgaard, 2023-02-11:
you need to test their api and look what you get back from it...
here are the raw requests

place replace the 3 Uppercase names with your own ids to test them.
There should be no private info in the output, but please check it before posting it
### Login
GET https://www.apsystemsema.com/ema/intoDemoUser.action?id=APSYSTEMS_AUTH

### Get Data
POST https://www.apsystemsema.com/ema/ajax/getReportApiAjax/getPowerOnCurrentDayAjax

queryDate=20230210&selectedValue=APSYSTEMS_ECU&systemId=APSYSTEMS_ID

rohrsh, 2023-02-12:
Am I doing this right? Sorry, not my forte. Via macOS terminal: 

curl -X GET "https://www.apsystemsema.com/ema/intoDemoUser.action?id=0b2848004bd780eb0........."

and

curl -X POST "https://www.apsystemsema.com/ema/ajax/getReportApiAjax/getPowerOnCurrentDayAjax" -d "queryDate=20230210&selectedValue=203000......&systemId=0b2848004bd780eb014bd88....."

Both give me a 403:

```
<html>
<head><title>403 Forbidden</title></head>
<body>
<center><h1>403 Forbidden</h1></center>
<hr><center>openresty/1.21.4.1</center>
</body>
</html>

```

skelgaard, 2023-02-12:
you get some headers with auth back from the first, you need in the second... i am unsure how you do that in curl, i do it with postman or http requests in phpstorm when i test stuff like this

rohrsh, 2023-02-12:
I turned on Home Assistant debugging logs. 

This was run at 10:18am local time in Australia. The last line is an odd time to reference. 

```
2023-02-13 10:18:08.108 DEBUG (MainThread) [custom_components.apsystems.sensor] post_data:
2023-02-13 10:18:08.108 DEBUG (MainThread) [custom_components.apsystems.sensor] {'queryDate': '20230213', 'selectedValue': '203000013419', 'systemId': '0b284800XXXXXXXXXXXXX'}
2023-02-13 10:18:08.108 DEBUG (MainThread) [custom_components.apsystems.sensor] starting: 2023-02-13 10:18:08
2023-02-13 10:18:09.818 DEBUG (MainThread) [custom_components.apsystems.sensor] status code data: 200
2023-02-13 10:18:09.818 DEBUG (MainThread) [custom_components.apsystems.sensor] 8.710357
2023-02-13 10:18:09.818 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:18:09.818 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:18:10.223 DEBUG (MainThread) [custom_components.apsystems.sensor] 0.407401
2023-02-13 10:18:10.223 DEBUG (MainThread) [custom_components.apsystems.sensor] 5696.0
2023-02-13 10:18:10.223 DEBUG (MainThread) [custom_components.apsystems.sensor] 5163
2023-02-13 10:18:10.223 DEBUG (MainThread) [custom_components.apsystems.sensor] 2023-02-13 19:41:39
```

Maybe this TODO matters for my situation?

[image]

I wonder if it is possible that the cache function is holding onto data for longer than it should?

rohrsh, 2023-02-13:
Here is the debug from the next few runs. There is no more "post_data" so I'm guessing it doesn't fetch from cloud again:

2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] 8.710357
2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] 0.407401
2023-02-13 10:23:10.226 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 5696.0
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 5163
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 2023-02-13 19:41:39
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:23:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 8.710357
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 0.407401
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 5696.0
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 5163
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.227 DEBUG (MainThread) [custom_components.apsystems.sensor] 2023-02-13 19:41:39
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:28:10.228 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:28:10.229 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
[... cut ...]
2023-02-13 10:58:10.233 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:58:10.233 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:58:10.233 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled
2023-02-13 10:58:10.233 DEBUG (MainThread) [custom_components.apsystems.sensor] Sunset variable: False
2023-02-13 10:58:10.233 DEBUG (MainThread) [custom_components.apsystems.sensor] Sensor is running. Sunset is disabled

skelgaard, 2023-02-13:
to se if it is the offset missing i need the output from the raw command... i'm showing that in debug.

rohrsh, 2023-02-17:
I commented out this line from sensor.py, to skip the cache:

`if (timestamp_now - timestamp_event > cache_time) and (timestamp_now - self.cache_timestamp > request_time):
`

And data seems to be updating properly now (e.g. for last two days)

[image]

Ganzhans, 2023-06-05:
I have a problem, I receive only an update when I restart ha ……
This is for the last 2 weeks, before it works great, has it to do with the update firmware off the site apsystemsemu,com ?

Ganzhans, 2023-06-06:
new behaviour  .... its updating without a restart once per hour instead of one per 5 minutes

skelgaard, 2023-06-07:
Something was updated on the ema portal... looking into it, but i'm not the developer of this... i have just fixed it to make it work for me...

Ganzhans, 2023-06-15:
> Something was updated on the ema portal... looking into it, but i'm not the developer of this... i have just fixed it to make it work for me...

ok, but have some new questions 

- you use is for your self 
- you have the same problems 
- may be I can help you with resolving it ( I am not a developer but some expiiance with scripts )
- the previous developer of this can may be help wit some hints 
- what / where is the trigger in HA to get new acutal data from the site apsystemsemu

ronaldvanvugt, 2023-06-15:
I have the same problem. Updating only every hour. I'll also willing to test a new version.

Ganzhans, 2023-06-16:
i think it has something to do with the cache., because when i restart Home assitant there is always new updated data with new timestamp
so can you explaine a litlte this part of the code "

`async def data(self):
        while self.running is True:
            await asyncio.sleep(1)

        if self.cache is None:
            await self.run()

        # continue None after run(), there is no data for this day
        if self.cache is None:
            _LOGGER.debug('No data')
            return self.cache
        else:
            # rules to check cache
            offset_hours = 7 * 60 * 60 * 1000
            timestamp_event = int(self.cache['time'][-1]) + offset_hours  # apsystems have 8h delayed in timestamp from UTC
            timestamp_now = int(round(time.time() * 1000))
            cache_time = 6 * 60 * 1000  # 6 minutes
            request_time = 20 * 1000  # 20 seconds to avoid request what is already requested

            if (timestamp_now - timestamp_event > cache_time) and (timestamp_now - self.cache_timestamp > request_time):
                await self.run()`

the status code "204" is also a point of disscision  maybe the code has afther te updated site an other number.  ?
hoe can i get data in a browser from Apsystemsema maby i can doe some test and come to a concluslion

Ganzhans, 2023-06-16:
i have changed this part 
` offset_hours = 5 * 60 * 60 * 1000`

because the sun is up more hours ,  and now i have a update every 5 minutes 

[image]

i don't now this is the right changed code, so i will testing to see or it works for a longer time.
i any one has a better solution for the cached code your welcome to share it and hopfully with comments

skelgaard, 2023-06-17:
well i found a few things that was coded bad....
and i fixed them
[image]

working on a few more small fixes, but will soon deploy a new version with thoose fixes in them

skelgaard, 2023-06-17:
pushed a new version with the new code in it, but not released it yet... do wanna run it a day or two before releaseing it

Ganzhans, 2023-06-18:
> pushed a new version with the new code in it, but not released it yet... do wanna run it a day or two before releaseing it

Thank you !  
I am running now 1.0.9. you here soon from me....

skelgaard, 2023-06-18:
this build also removed the need for extra code in configuration.yml to make it usable in the energy dashboard, so if you have that, you can remove it

Ganzhans, 2023-06-18:
> this build also removed the need for extra code in configuration.yml to make it usable in the energy dashboard, so if you have that, you can remove it

What is wrong with the energy dashboard ? For what I know there’s no problem with mine….

skelgaard, 2023-06-18:
> > this build also removed the need for extra code in configuration.yml to make it usable in the energy dashboard, so if you have that, you can remove it
> 
> What is wrong with the energy dashboard ? For what I know there’s no problem with mine….

no problem, just that if you added extra code to make it show up, thats not nedded anymore... i fixed it so it now shows up by it selves.
Talking about the code talked about in https://github.com/skelgaard/homeassistant-apsystems/issues/20

skelgaard, 2023-06-19:
i will close this issue, please create a new one, if you run into more problems