skelgaard/homeassistant-apsystems #31: times the sensor is online and generate data

Ganzhans, 2023-07-05:
I have found a problem, but nt sure ist  problem in version 1.0.9 

this is what happend ......
1 ) there is a little gap between around sunset  so the last update of that day is not grabbed because the converter is offline at that time ......
2) the next day the convert becomes online the last update is notice by step 1  wil be registered (and counted on that day)

look the 2 pictures 

[image]

 
[image]

how can is solve this  ? I have this tought about this 

1) adjust the interval of 5 min  to 1 min 1 hour before sun set
2) update he interval the whole day to 1 min
3) try to update between sunrise - 1 hour and on sunset + 1 hour

is this possible ?

skelgaard, 2023-07-05:
first just turn off sunset ... set it to off.
if it is fixed by that, then yes the other sollution will fix it to.

Ganzhans, 2023-07-08:
I set sunset to off, works great !!

shame on me ............

skelgaard, 2023-07-08:
i will se, if i can implement the other suggestion, but as long as it works that way :)
or i may just remove the sunet in all... doesn't really matter that much when it only goes with 1 every min :)

closed 2023-09-01 by skelgaard