skelgaard/homeassistant-apsystems #32: APSystem device and sensors not visible

schalkk, 2023-09-14:
I see the following in my log file:

2023-09-14 12:15:25.111 DEBUG (MainThread) [custom_components.apsystems.sensor] post_data:                     
2023-09-14 12:15:25.111 DEBUG (MainThread) [custom_components.apsystems.sensor] {'queryDate': '20230914', 'sele
ctedValue': 'xxxxxxx', 'systemId': 'yyyyyyy'}                                    
2023-09-14 12:15:25.111 DEBUG (MainThread) [custom_components.apsystems.sensor] starting: 2023-09-14 12:15:25  
2023-09-14 12:15:25.260 DEBUG (MainThread) [custom_components.apsystems.sensor] status code data: 200          
2023-09-14 12:15:25.260 ERROR (MainThread) [homeassistant.components.sensor] apsystems: Error on device update!
  File "/config/custom_components/apsystems/sensor.py", line 209, in async_update                              
  File "/config/custom_components/apsystems/sensor.py", line 313, in data                                      
  File "/config/custom_components/apsystems/sensor.py", line 301, in run 

Could you point me where to look?

skelgaard, 2023-09-14:
can you login to the webinterface apsystemsema.com ?

schalkk, 2023-09-14:
Yes!

schalkk, 2023-09-14:
Which URL should I use with the curl command to see that my Docker can access?

skelgaard, 2023-09-14:
### Login
GET https://www.apsystemsema.com/ema/intoDemoUser.action?id=apsystemsem_authid

### Get Data
POST https://www.apsystemsema.com/ema/ajax/getReportApiAjax/getPowerOnCurrentDayAjax

queryDate=20230615&selectedValue=apsystemsema_ecu_id&systemId=apsystemsema_system_id

schalkk, 2023-09-14:
Do I need to pass a sessionID or something from Login to GET DATA? (using PostMan to test with before trying curl on Unix.)

schalkk, 2023-09-14:
SessionID is now passed via Cookies from Login.  Get data states:

<div class="flex" style="height: 100%;">
		<div class="title-text">Note</div>
		<div class="info-text">We are so sorry, EMA has encountered an
			error.<br/>click [OK] to email us and we will deal with it within 48 hours. Please re-login your system.</div>

			<div>

skelgaard, 2023-09-14:
so an EMA error... .. can't do much other than catch the error and give an error in debug about it instead,,,,

schalkk, 2023-09-14:
Will this cause the device and sensors not to show up?

joceliodownloads, 2023-09-30:
Any solution?

skelgaard, 2023-09-30:
mine works fine.... 
can you se data in their interface and can you fetch data by the manual functions ?
if you can't fetch data by the manual function then your phone app shouldn't work either as its same way it fetches data on.

closed 2024-02-03 by skelgaard