skelgaard/homeassistant-apsystems #35: Energy section not seeing Solar until after 3pm

gwilliams1205, 2024-03-18:
Over the last 2 weeks something appears to have changed in the way the system does polling.. My sensors are now staying dormant until around 3pm every day.. Then just showing what has been produced until then. I thought it was the sunset setting, but I changed that to off with no difference.
[image]

The sensors also shows the wrong date until after 3pm again:

[image]

Please help

skelgaard, 2024-03-18:
whats is your timezone ?

you need to test their api and look what you get back from it...
here are the raw requests

place replace the 3 Uppercase names with your own ids to test them.
There should be no private info in the output, but please check it before posting it
Login

GET https://www.apsystemsema.com/ema/intoDemoUser.action?id=APSYSTEMS_AUTH
Get Data

POST https://www.apsystemsema.com/ema/ajax/getReportApiAjax/getPowerOnCurrentDayAjax

queryDate=20240317&selectedValue=APSYSTEMS_ECU&systemId=APSYSTEMS_ID

skelgaard, 2024-03-18:
well that means their API is down and then it doesn't work... it will then pick up the "slack" when it comes up
does https://apsystemsema.com work for you ?

gwilliams1205, 2024-03-18:
It was working fine then something changed about 2 weeks ago.. Have you not seen any changes?

skelgaard, 2024-03-18:
well i can't say then.... if you look at network traffic when going to their site, its the same url with the same parameters i call.
no mine is working fine.
This is mine from today
[image]

skelgaard, 2024-03-18:
btw you need todo thoose post in postman or something like that, as it needs info from the first in the second call

gwilliams1205, 2024-03-18:
I am using postman.. is there any special setting I need set to make sure it's looking at the response from the GET when sending the POST?

skelgaard, 2024-03-18:
nope that should work fine then, but the problem is the connection to their API... its always that when it is and as long as EMA don't let us fetch that data direct from their router, then there is no other way to get the data, unless you get a shelly or something else to messure it.

gwilliams1205, 2024-03-18:
well that's less than Ideal..

Thanks for the help though.. It just stinks it was working great then decided to stop all of a sudden

skelgaard, 2024-03-18:
i wish there was something i could do, but if your not able to pull data from their api, then neither can home assistant and that will give the graph you showed.

closed 2024-03-18 by skelgaard