skelgaard/homeassistant-apsystems #37: APSystems website updated, SystemID not to be found

matthijsvp, 2024-06-06:
Hi,

First of all thanks for building this integration! I'd like to try it, however, the APSystems website seems to have changed since you wrote the README on how to find the systemid.

This: 

> <span>Settings</span>
> <ul>
>     <li onclick="managementClickCustomer('YOUR SYSTEM ID')"><a>Settings</a></li>
>     <li onclick="intoFaq(10)"><a>Help</a></li>
> </ul>

Is nowhere to be found. Would you perhaps be able to help out and see if there's a new way to find it? I can find the Auth ID and the ECU ID.

skelgaard, 2024-06-06:
i need to update the readme, if you go into settings and set this checkmark, the id comes up
[image]

matthijsvp, 2024-06-06:
Isn’t that the Auth ID?

skelgaard, 2024-06-06:
the other id is still in the souce
<li class="nav-menu nav-h-managementcustomer"><a id="managementcustomer_head" class="onselect_a" onclick="managementClickCustomer('XXXXXXX')"><span class="onselect_span"></span>SETTING</a></li>
just press ctrl+u (or how to view source in your browser) on frontpage and search for managementClickCustomer( and it will be there

matthijsvp, 2024-06-06:
Ahhhh gotcha! Thought I had to do that on the Settings page. Thanks!

skelgaard, 2024-06-06:
np

closed 2024-06-06 by skelgaard