skelgaard/homeassistant-apsystems #38: Struggling with login, getting sensors to show

tmontgom, 2024-08-07:
Hi, I'm new to home assistant and command line/linux in general so I might be missing something obvious. But I can't get the two test commands to give good results and there are no sensors or devices or anything showing up for apsystems that I can tell.

I installed the HA community store and fed it the github url through the custom repository option. I added the tiny bit of code to the configuration.yaml file and changed the values to my numbers, so it looks like:

```
sensor:
- platform: apsystems
- authId: "Auth id I got from apsystemsema.com here"
- systemId: "system id I got from apsystemsema.com here"
- ecuId: "ECU id I got from apsystemsema.com here"
- sunset: off
```

I also added the logger code to the configuration.yaml file with no changes.

`grep apsystem home-assistant.log`
doesn't work but
`grep apsystem /config/home-assistant.log`
gives me
`2024-08-07 11:02:23.838 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration apsystems which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.`

The "get" command isn't recognized but
`curl https://www.apsystemsema.com/ema/intoDemoUser.action?id=apsystemsem_authid`
provides the html for a 403 Forbidden error. Similarly "post" wasn't recognized but
`curl https://www.apsystemsema.com/ema/ajax/getReportApiAjax/getPowerOnCurrentDayAjax`
returns the same 403 forbidden error.

I'm assuming they aren't recognized because this is basically a fresh install of the Home Assistant OS on a Raspberry Pi.

The init, manifest, and sensor files are in the custom components folder. In the settings, there is nothing that looks apsystems-related under Integrations, Devices, or Entities.

Any advice? Thanks for your time in keeping all this running!

skelgaard, 2024-08-08:
if you don't get any data, start by settting the debugger on and se what else it shows in the log in home assistant or by the grep cmmand to se if you can find out why you get no data

tmontgom, 2024-08-09:
It seems to have fixed itself after a reboot. It's possible I just never restarted it in the first place so the changes weren't being picked up. Thanks for your time!

closed 2024-08-09 by tmontgom