skelgaard/homeassistant-apsystems #39: Errors in log file

wimsan, 2024-09-23:
I have configured the plugin. I also set logging to debug.
Now I constantly see errors like this in teh log:

Logger: homeassistant.components.sensor
Bron: components/sensor/__init__.py:569
integratie: Sensor (documentatie, problemen)
Eerst voorgekomen: 19 september 2024 om 14:38:20 (1 gebeurtenissen)
Laatst gelogd: 19 september 2024 om 14:38:20

Entity sensor.apsystems_energy_latest (<class 'custom_components.apsystems.sensor.ApsystemsSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'apsystems' custom integration

wimsan, 2024-09-23:
in home-assitant.log I see entries like this:

2024-09-23 15:40:29.521 ERROR (MainThread) [custom_components.apsystems.sensor] 'buyTotal' => ap_data : {'duration': 48, 'total': '7.573157', 'max': '2352.0', 'F': [0, 0, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1], 'co2': '7.550438', 'time': [1727047603000, 1727048204000, 1727048806000, 1727049412000, 1727050017000, 1727050619000, 1727051222000, 1727051822000, 1727052422000, 1727053023000, 1727053628000, 1727054232000, 1727054834000, 1727055439000, 1727056039000, 1727056643000, 1727057242000, 1727057846000, 1727058446000, 1727059051000, 1727059653000, 1727060256000, 1727060859000, 1727061461000, 1727062065000, 1727062671000, 1727063275000, 1727063877000, 1727064478000, 1727065079000, 1727065681000, 1727066284000, 1727066887000, 1727067491000, 1727068093000, 1727068698000, 1727069302000, 1727069905000, 1727070507000, 1727071109000, 1727071711000, 1727072313000, 1727072915000, 1727073519000, 1727074122000, 1727074727000, 1727075332000, 1727075934000, 1727076539000], 'power': ['20', '25', '39', '43', '37', '41', '45', '44', '42', '128', '200', '235', '411', '721', '604', '480', '570', '845', '1119', '1143', '705', '717', '567', '318', '315', '1537', '565', '1626', '1789', '1980', '1971', '2000', '2136', '2352', '2069', '2083', '2014', '2076', '779', '1829', '1400', '1372', '1073', '1460', '1631', '927', '1141', '1437', '708'], 'energy': ['0.003196', '0.003999', '0.006233', '0.006876', '0.005915', '0.006558', '0.007194', '0.007035', '0.006717', '0.020463', '0.031974', '0.037570', '0.065712', '0.115271', '0.096565', '0.076739', '0.091128', '0.135098', '0.178901', '0.182740', '0.112711', '0.114627', '0.090651', '0.050839', '0.050361', '0.245726', '0.090333', '0.259956', '0.286015', '0.316550', '0.315111', '0.319754', '0.341496', '0.376030', '0.330780', '0.333022', '0.321988', '0.331901', '0.124541', '0.292415', '0.223825', '0.219349', '0.171547', '0.233420', '0.260759', '0.148207', '0.182415', '0.229739', '0.113188'], '0b28482f7849305d0178497d9fa51087': ['3', '4', '5', '6', '5', '5', '6', '6', '5', '13', '20', '23', '41', '71', '58', '46', '57', '100', '130', '127', '69', '71', '57', '33', '34', '173', '61', '175', '190', '211', '207', '206', '214', '229', '200', '199', '191', '194', '73', '164', '125', '121', '94', '127', '136', '78', '96', '119', '64'], '0b28482f7849305d0178497d9fa5107f': ['3', '3', '4', '5', '4', '5', '5', '5', '5', '12', '17', '20', '32', '61', '50', '40', '50', '89', '114', '109', '58', '61', '49', '28', '25', '149', '50', '148', '161', '179', '176', '175', '177', '192', '168', '167', '160', '162', '62', '139', '104', '102', '80', '107', '114', '65', '81', '100', '54'], '0b28482f7849305d0178497d9fa51086': ['3', '4', '5', '6', '5', '5', '6', '6', '6', '13', '21', '25', '40', '63', '55', '44', '49', '54', '74', '83', '63', '62', '49', '30', '30', '106', '47', '123', '137', '153', '155', '162', '181', '205', '183', '187', '183', '192', '71', '174', '136', '134', '104', '145', '167', '95', '116', '148', '68'], '0b28482f7849305d0178497d9fa5107e': ['4', '5', '6', '7', '6', '7', '8', '7', '8', '15', '20', '24', '42', '72', '58', '47', '58', '102', '132', '128', '69', '71', '57', '34', '35', '179', '62', '177', '194', '214', '210', '209', '216', '230', '202', '200', '192', '195', '74', '164', '124', '121', '94', '127', '136', '78', '96', '119', '64'], '0b28482f7849305d0178497d9fa51085': ['0', '0', '1', '2', '1', '1', '1', '2', '1', '8', '15', '16', '32', '58', '50', '40', '45', '49', '69', '79', '58', '58', '45', '23', '20', '101', '42', '119', '131', '147', '149', '156', '175', '199', '177', '180', '176', '186', '67', '168', '131', '128', '99', '139', '161', '90', '110', '142', '64'], '0b28482f7849305d0178497d9fa5107d': ['1', '1', '2', '3', '2', '3', '4', '3', '4', '12', '18', '22', '38', '60', '52', '41', '47', '51', '72', '81', '60', '60', '47', '27', '28', '104', '44', '121', '134', '151', '153', '159', '178', '204', '180', '184', '180', '189', '69', '172', '133', '131', '103', '141', '164', '93', '113', '145', '66'], '0b28482f7849305d0178497d9fa51084': ['0', '0', '1', '2', '1', '1', '2', '2', '1', '9', '16', '15', '33', '67', '54', '42', '53', '97', '125', '121', '64', '67', '52', '26', '25', '169', '56', '170', '186', '206', '202', '201', '208', '222', '194', '193', '185', '187', '69', '157', '118', '114', '88', '121', '129', '73', '90', '113', '59'], '0b28482f7849305d0178497d9fa5107c': ['0', '0', '2', '2', '2', '2', '2', '2', '2', '8', '13', '16', '28', '49', '43', '34', '37', '42', '59', '67', '50', '49', '38', '20', '21', '85', '34', '99', '110', '121', '125', '130', '142', '164', '145', '148', '144', '152', '57', '139', '108', '106', '84', '115', '133', '75', '92', '118', '54'], '0b28482f7849305d0178497d9fa51083': ['3', '3', '4', '4', '4', '5', '4', '4', '4', '12', '18', '22', '34', '53', '46', '37', '42', '46', '63', '70', '53', '52', '42', '25', '26', '91', '39', '104', '116', '128', '130', '135', '150', '169', '150', '153', '149', '156', '60', '144', '112', '111', '88', '118', '137', '79', '97', '122', '57'], '0b28482f7849305d0178497d9fa51082': ['3', '4', '5', '4', '4', '4', '4', '4', '4', '11', '16',  [... cut ...] , '145', '46', '143', '158', '172', '168', '167', '173', '185', '160', '160', '153', '154', '58', '132', '99', '97', '76', '101', '108', '61', '77', '95', '50'], '0b28482f7849305d0178497d9fa51080': ['0', '1', '2', '1', '2', '2', '2', '2', '1', '8', '14', '17', '30', '50', '44', '34', '38', '42', '60', '67', '50', '50', '38', '21', '21', '86', '34', '100', '111', '123', '125', '130', '145', '165', '146', '149', '145', '152', '58', '140', '108', '107', '85', '115', '134', '76', '93', '118', '55']}

skelgaard, 2024-09-25:
well i have been testing and only see errors like this, if apsystem site is down/unreacable else i see no problem... please post if you have more information and i will look on into it

closed 2024-09-25 by skelgaard

wimsan, 2024-09-30:
I now use the integration from https://github.com/jeremletrol81/homeassistant-apsystems. and I see this warninngin the log:

2024-09-26 15:29:35.348 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.apsystems_energy_latest (<class 'custom_components.apsystems.sensor.ApsystemsSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'apsystems' custom integration

skelgaard, 2024-09-30:
well then takl to him about it...
i don't understand why instead of making a new version, he doesn't do a merge request so that only one working version is out there instead of 2 different versions.....