skelgaard/homeassistant-apsystems #40: Rename integration to not conflict with HA core apsystems integration

fcrozat, 2024-12-17:
Since version 2024.6, HA has a APSystems integration (which is quering directly micro-inverters) and "complains" this extension is shadowing the "core" one.

It would be a good idea to rename this extension to prevent this conflict.

skelgaard, 2024-12-17:
First off it should be theirs that renamed theirs as is new and this one has been there long.
Second of, you should only use one of them, there should newer be a need for both.

Now that said, i will look into if i can easy do this and what it will do or it will break everything for everyone....

skelgaard, 2024-12-29:
i have renamed and tested, but i have not pushed yet.. this is cause it will break everything you have setup, as when it changes, it will change the device id and there for all places you have set it in, in graphs and others will have to be fixed... i will push it later though

Walker140906, 2025-03-29:
Hey ! what about now ? Did you rename it ?

skelgaard, 2025-09-05:
will be renamed in next release this weekend or next week

closed 2025-09-05 by skelgaard