skelgaard/homeassistant-apsystems #44: Daily blockage

JanSchreurs, 2025-09-03:
It seems that every morning access to Apsystems data is blocked. I can unblock it by manually logging in to my APsystems account on their website and resaving the setttings file. The moment I do that I can see that data are received again.  My conclusion is that something has changed at the APsystems side wich blocks access. The integration worked fine until a few weeks back.  I have reported this as well to APsystems, but their response has been all but helpfull. Could it be that when the ECU is switched off at night by APsystems it still gets requests for access from Home Assistant and after a few of these blocks the account until I manually unblock? Any idea what I need to do to resolve this problem?

JanSchreurs, 2025-09-13:
> new version will be released this weekend or next week

Hi I have noticed there is a new release, but it does not solve the problem that I reported before. APsystems getting locked overnight and only getting data after manually logging in as Apsystems user. Which means one needs to login on a daily basis. GreetingsJanSchreurs ***@***.***

skelgaard, 2025-09-13:
do you have more instanses of homeassistant running or other software ? 
if you need to manuelly login with captch its not a api issue... its something else that get you captcha banned and that can't be handle on the api... that needs a manual login to fix.

JanSchreurs, 2025-09-14:
Only one installed in the home, but I do access home assistant as well via the home assistant app on my ipad. That’s a fitting expression, getting ‘captcha banned’. Did not know that could happen. In APsystems app I see and Open API and I wonder whether there is something to change there. It is active since august, but there statistics listed do not show any open api access activity.GreetingsJanSchreurs ***@***.***

if you need to manuelly login with captcha its not a api issue... its something else that get you captcha banned and that can't be handle on the api... that needs a manual login to fix.

skelgaard, 2025-09-14:
how much you acccess HA doesn't matter... HA does 1 call a min to the api for the data

if you want to increase that, there is a setting in the sensor.py file
SCAN_INTERVAL = timedelta(minutes=1)

that can be setup to higher... i can add that to config, so you can change that dynamic in a new version.
but other than that, i can't do anything... that their API works on the same captcha db as the normal login, is just bad design.

skelgaard, 2025-09-14:
when you set debug on, you can see in the log what it does.

so enable debug and restart and look in logs what it does. paste them here if it is...

JanSchreurs, 2025-09-14:
Many thanks,With your explanation in your last sentence I will contact APsystems. GreetingsJanSchreurs ***@***.***

skelgaard, 2025-09-14:
its unofficial that you can use their api that way, they will proberly not support it at all so don't think you will get anywhere with it.

JanSchreurs, 2025-10-20:
Apologies for late reply, I have been away from home. No the problem is still the same. My solution is logging in manually each morning. It is a pain!

skelgaard, 2025-10-20:
@CyrilleBonamy i can't do anything about it, its not possible to fix though the api's i have found, so there is nothing i can do...
i can make update interval configurable if setting the SCAN_INTERVAL to 5 in the file helps, but else there is nothing i can do